monemati/PX4-ROS2-Gazebo-YOLOv8 #1: About execution issue

JustinHsuNctu, 2024-01-19:
Hello! I try to follow your project. Your projrct is good for my research topic. My environment is Ubuntu 22.04(jammy). However, when I try to install PX4, there is an error for protoc version. I check the protoc version and found that the version is 3.12. Therefore, I change the environment to Ubuntu 20.04 (Focal). The PX4 complier is OK and the protoc version is 3.6.1. But in Ubuntu 20.04, ROS2 Humble could not be installed! Could you give me a suggestion how should I do?
Thanks much!!
Justin Hsu

monemati, 2024-01-19:
@JustinHsuNctu Hello! Thank you for using my repo. I suggest you to stick to Ubuntu 22 and try do resolve protoc issue, but if you want to continue on Ubuntu 20, you can build Humble on it. There is a guide in ROS2 documentation.
https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

JustinHsuNctu, 2024-01-26:
@monemati Thank you for your fast reply! Now, I move on to Ubuntu 22.04. The protobuf problem is still there. My error message are shown in below.
I try to downgrade my protobuf to 3.6.1 and 2.6.1, the error is still there.

By the way, In Ubuntu 20.04, I had been successed compiler all of your toolchain. However, when I execute  
Terminal #3:
ros2 run ros_gz_image image_bridge /camera
I found that ros2 foxy version do not support ros_gz_image. So I need to move on to Ubuntu 22.04 to install ros2 humble. But protobuf problem is still there.
Would you like to give me some suggestion? 
Justin Hsu
 

/usr/include/gz/msgs9/gz/msgs/details/pointcloud_packed.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
[674/967] Building CXX object src/modules/simulation/gz_br...es__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o
/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"gz_bridge\" -DPROTOBUF_USE_DLLS -DPX4_MAIN=gz_bridge_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/home/ubuntu/PX4-Autopilot/boards/px4/sitl/src -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/common/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/src/lib -I/home/ubuntu/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/home/ubuntu/PX4-Autopilot/platforms/common -I/home/ubuntu/PX4-Autopilot/platforms/common/include -I/home/ubuntu/PX4-Autopilot/src -I/home/ubuntu/PX4-Autopilot/src/include -I/home/ubuntu/PX4-Autopilot/src/lib -I/home/ubuntu/PX4-Autopilot/src/lib/matrix -I/home/ubuntu/PX4-Autopilot/src/modules -I/home/ubuntu/PX4-Autopilot/platforms/posix/include -I/home/ubuntu/PX4-Autopilot/build/px4_sitl_default/external/Install/include -isystem /usr/include/gz/transport12 -isystem /mnt/c/ProgramData/anaconda3/Library/include -isystem /usr/include/gz/utils2 -isystem /usr/include/gz/msgs9 -isystem /usr/include/gz/math7 -isystem /usr/include/uuid -O2 -g -DNDEBUG -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -Wno-stringop-truncation -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -Wreorder -Wno-overloaded-virtual -frtti -fexceptions -O2 -I/usr/include/uuid -std=gnu++17 -MD -MT src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o -MF src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o.d -o src/modules/simulation/gz_bridge/CMakeFiles/modules__simulation__gz_bridge.dir/GZMixingInterfaceServo.cpp.o -c /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp
In file included from /usr/include/gz/msgs9/gz/msgs/discovery.pb.h:17,
                 from /usr/include/gz/transport12/gz/transport/Discovery.hh:59,
                 from /usr/include/gz/transport12/gz/transport.hh:26,
                 from /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.hpp:38,
                 from /home/ubuntu/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp:34:
/usr/include/gz/msgs9/gz/msgs/details/discovery.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors.
[676/967] Building CXX object src/modules/replay/CMakeFiles/modules__replay.dir/Replay.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1

JustinHsuNctu, 2024-01-28:
@monemati Thank you for your fast reply! I try a different way. I install protobuf 2.6.1, then I remove protobuf-compiler ver 3,12,4 by apt remove. The compiler of PX4-Autopilot has been passed, and all the toolchains that you mentioned have been installed. But when I execute 
PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4
Other error messages are shown in below.
I copied models from main repo to ~/.gz/models, also copied default.sdf from worlds folder in the main repo to ~/PX4-Autopilot/Tools/simulation/gz/worlds/. But when I want to " Change the angle of Drone's camera for better visual," I could not found the folder "~/PX4-Autopilot/Tools/simulation/gz/models/x500_depth/"
Do I miss any step or something wrong?
Justin Hsu

The ERROR message: 

  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
ERROR [param] Parameter SIM_GZ_EN not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC1 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC2 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC3 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC4 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN1 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN2 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN3 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN4 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX1 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX2 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX3 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX4 not found.
INFO  [dataman] data manager file './dataman' size is 7868392 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /default.sdf
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 268.08,-128.22,3.86,0.00,0,-0.7
etc/init.d-posix/rcS: 123: /home/ubuntu/PX4-Autopilot/build/px4_sitl_default/rootfs/etc/init.d-posix/px4-rc.simulator: gz_bridge: not found
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
ubuntu@LAPTOP-RJH1RO1J:~/PX4-Autopilot$ [Err] [SystemPaths.cc:534] File [/default.sdf] resolved to path [/default.sdf] but the path does not exist
[Err] [Server.cc:109] Failed to find world [/default.sdf]

monemati, 2024-01-28:
@JustinHsuNctu I don't know if u missed a step while compiling PX4 or not but I think the errors are because PX4 can not find the models and worlds directories, as you can see it says
"INFO [init] starting gazebo with world: /default.sdf"
which is not a good sign since this is from your root directory. It should be something like this:
"INFO [init] starting gazebo with world: /home/ubuntu/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf"

JustinHsuNctu, 2024-01-29:
@monemati I am not very sure! I posted a bug report on the PX4-Autopilot's repo. May be they have some comment about it.
If they have some resolution, I will tell you!
Justin Hsu

JustinHsuNctu, 2024-01-29:
I found the solution! In WSL2, all commands should execute in Windows PowerShell environment.
It is described on PX4 official website.
"All operations to install and build PX4 must be done within a WSL Shell (you can use the same shell that was used to install WSL2 or open a new one)."
from "https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html"

monemati, 2024-01-30:
@JustinHsuNctu Oh I didn't know you are using WSL, nice find though, I am glad it is resolved. Enjoy your flight 🚁

closed 2024-05-11 by monemati