monemati/PX4-ROS2-Gazebo-YOLOv8 #2: About keboard control and image capture problem

JustinHsuNctu, 2024-01-31:
@monemati  Hello, it's me, again! Sorry for bother! Now, I build all environment. I execute the keboard control part. As you mentioned, I open 5 terminal. When I execute the 5th terminal ( execute python keyboard-mavsdk-test.py) and try to control the UAV, The "detected frame" window does not chnage. There is no message in Terminal 3 (execute ros2 run ros_gz_image image_bridge /camera). 
When I press key AWSD, there are some value changed in terminal 5, the third value become 0.5 or 1 or -0.5 etc.
Could you give me some suggestion help me to debug it?
Justin Hsu

The messages of the five terminals are shown in below.
The message in terminal 1 are:
[1706686107.313128] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1706686107.313734] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1706686151.732744] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1706686151.733181] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 127.0.0.1:48042
[1706686151.739228] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1706686151.739605] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x800(2), participant_id: 0x001(1)
[1706686151.739723] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x800(4), participant_id: 0x001(1)
[1706686151.741238] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x800(6), subscriber_id: 0x800(4)
[1706686151.741537] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x801(2), participant_id: 0x001(1)
[1706686151.741596] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x801(4), participant_id: 0x001(1)
[1706686151.741795] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x801(6), subscriber_id: 0x801(4)
[1706686151.742379] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x802(2), participant_id: 0x001(1)
[1706686151.742452] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x802(4), participant_id: 0x001(1)
[1706686151.742677] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x802(6), subscriber_id: 0x802(4)
[1706686151.742973] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x803(2), participant_id: 0x001(1)
[1706686151.743036] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x803(4), participant_id: 0x001(1)
[1706686151.743248] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x803(6), subscriber_id: 0x803(4)
[1706686151.743521] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x804(2), participant_id: 0x001(1)
[1706686151.743597] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x804(4), participant_id: 0x001(1)
[1706686151.743827] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x804(6), subscriber_id: 0x804(4)
[1706686151.744081] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x805(2), participant_id: 0x001(1)
[1706686151.744142] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x805(4), participant_id: 0x001(1)
[1706686151.744345] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x805(6), subscriber_id: 0x805(4)
[1706686151.744623] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x806(2), participant_id: 0x001(1)
[1706686151.744683] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x806(4), participant_id: 0x001(1)
[1706686151.744871] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x806(6), subscriber_id: 0x806(4)
[1706686151.745084] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x807(2), participant_id: 0x001(1)
[1706686151.745139] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x807(4), participant_id: 0x001(1)
[1706686151.745374] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x807(6), subscriber_id: 0x807(4)
[1706686151.745577] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x808(2), participant_id: 0x001(1)
[1706686151.745631] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x808(4), participant_id: 0x001(1)
[1706686151.745823] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x808(6), subscriber_id: 0x808(4)
[1706686151.746307] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x809(2), participant_id: 0x001(1)
[1706686151.746369] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x809(4), participant_id: 0x001(1)
[1706686151.746576] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x809(6), subscriber_id: 0x809(4)
[1706686151.746869] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80A(2), participant_id: 0x001(1)
[1706686151.746930] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80A(4), participant_id: 0x001(1)
[1706686151.747184] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80A(6), subscriber_id: 0x80A(4)
[1706686151.747471] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80B(2), participant_id: 0x001(1)
[1706686151.747527] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80B(4), participant_id: 0x001(1)
[1706686151.747746] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80B(6), subscriber_id: 0x80B(4)
[1706686151.747963] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80C(2), participant_id: 0x001(1)
[1706686151.748020] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80C(4), participant_id: 0x001(1)
[1706686151.748192] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80C(6), subscriber_id: 0x80C(4)
[1706686151.748432] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80D(2), participant_id: 0x001(1)
[... cut ...]
[1706686151.751295] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x812(6), subscriber_id: 0x812(4)
[1706686151.751544] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x813(2), participant_id: 0x001(1)
[... 576 lines cut ...]
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0
0 0 0.5 0

monemati, 2024-01-31:
@JustinHsuNctu Hi Justin. Terminal #3 is not verbose and won't show any messages so it is ok.
Before hitting WASD you should hit "r" to arm the drone.
Oh I can see this error in your terminal: raise ActionError(result, "arm()"). Check if the drone is in a good position to arm and also check if gps is fixed (wait about 15-30 sec then hit "r").
And also if you are in WSL check if everything is running in WSL and your environment is ok.

JustinHsuNctu, 2024-02-01:
@monemati Thank you for your reply! But, How could I check if gps is fixed? And how could I know if the drone is in a good postition? Does any message that I can see?

JustinHsuNctu, 2024-02-01:
Now, I can control UAV flying. But, the takeoff action should give it by "internal command". That means I should give the command in treinal 2 "pxh>commander takeoff." After that, I can use keyboard to control the UAV. 
By the way, the delay of controlling is very very long!

monemati, 2024-02-02:
> @monemati Thank you for your reply! But, How could I check if gps is fixed? And how could I know if the drone is in a good postition? Does any message that I can see? Also, I provide the terminal 2 messages.
> 
> Terminal 2:
> 
> INFO [logger] Opened full log file: ./log/2024-02-01/06_44_04.ulg INFO [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network) INFO [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network) INFO [px4] Startup script returned successfully pxh> INFO [commander] Ready for takeoff! INFO [uxrce_dds_client] synchronized with time offset 1706769844356401us INFO [uxrce_dds_client] successfully created rt/fmu/out/battery_status data writer, topic id: 19 INFO [uxrce_dds_client] successfully created rt/fmu/out/estimator_status_flags data writer, topic id: 86 INFO [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 92 INFO [uxrce_dds_client] successfully created rt/fmu/out/manual_control_setpoint data writer, topic id: 138 INFO [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 169INFO [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 191 INFO [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 214 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 230 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 237 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 238 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_gps_position data writer, topic id: 240 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 244 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 249 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 254 WARN [timesync] time jump detected. Resetting time synchroniser. INFO [mavlink] partner IP: 127.0.0.1 INFO [uxrce_dds_client] successfully created rt/fmu/out/vehicle_command_ack data writer, topic id: 234 WARN [timesync] time jump detected. Resetting time synchroniser. INFO [commander] Armed by external command INFO [tone_alarm] arming warning INFO [commander] Pilot took over using sticks INFO [tone_alarm] notify positive INFO [commander] Takeoff detected WARN [timesync] time jump detected. Resetting time synchroniser. WARN [timesync] time jump detected. Resetting time synchroniser. Stack trace (most recent call last) in thread 10200: #10 Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in #9 Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc33729884f, in #8 Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc337206ac2, in #7 Object "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so", at 0x7fc30e2957ca, in #6 Object "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so", at 0x7fc30e245bc7, in #5 Object "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so", at 0x7fc30e7cf611, in #4 Object "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so", at 0x7fc30e7d4547, in #3 Object "/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so", at 0x7fc30ece311a, in #2 Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7fc320d473af, in #1 Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7fc320d3383a, in #0 Object "/usr/lib/wsl/lib/libd3d12core.so", at 0x7fc320dd9a60, in Segmentation fault (Address not mapped to object [0x180])

If your drone is in a good position on the ground (not tilted and not skewed) it will get gps fixed in the first minute of booting up.

monemati, 2024-02-02:
> Now, I can control UAV flying. But, the takeoff action should give it by "internal command". That means I should give the command in treinal 2 "pxh>commander takeoff." After that, I can use keyboard to control the UAV. By the way, the delay of controlling is very very long!

I ran it again on my machine and it does not need additional command and there is no delay, maybe it is because you are running it on WSL. Btw I am using ubuntu on my machine as an OS not vmware or WSL. Do you have access to a machine with ubuntu OS?

JustinHsuNctu, 2024-02-04:
> I ran it again on my machine and it does not need additional command and there is no delay, maybe it is because you are running it on WSL. Btw I am using ubuntu on my machine as an OS not vmware or [...]

Thank you for your suggestion. I will try it.

JustinHsuNctu, 2024-04-11:
@monemati Hi, hello, sorry, I forgot to update my testing result. I follow your suggestion to move on to pure Ubuntu environment. Everything is OKay now! I can build up my own environment with Gazebo simulation. Thnak you all your help!
Justin Hsu

monemati, 2024-04-11:
@JustinHsuNctu Hi, thank you for the update, have a happy flight 🚀

closed 2024-05-11 by monemati