monemati/PX4-ROS2-Gazebo-YOLOv8 #4: Error Gazebo -  PX4-ROS2-Gazebo-YOLOv8

Mojgan-arch, 2024-03-03:
**Hi, I hope you are well. I have problem to see Gazebo scene after installing [PX4-ROS2-Gazebo-YOLOv8], Gazebo page appear but completely black. I have ROS2 Humble on my pc. I installed ROS2 before, so I did not just follow that part related to install Ros2, but I did all process step by step.**

**When I run this code in Terminal, Gazebo page appear in black scene.** 
(Terminal #2:
cd ~/PX4-Autopilot
PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4)

**Would you please guide me to solve my problem? Thanks in advance**

This is my Error: 
desktop:~/PX4-Autopilot$ cd ~/PX4-Autopilot
desktop:~/PX4-Autopilot$ PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

______         ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 4002
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
ERROR [param] Parameter SIM_GZ_EN not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC1 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC2 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC3 not found.
ERROR [param] Parameter SIM_GZ_EC_FUNC4 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN1 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN2 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN3 not found.
ERROR [param] Parameter SIM_GZ_EC_MIN4 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX1 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX2 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX3 not found.
ERROR [param] Parameter SIM_GZ_EC_MAX4 not found.
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /default.sdf
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 268.08,-128.22,3.86,0.00,0,-0.7
etc/init.d-posix/rcS: 117: /home/Desktop/PX4-Autopilot/build/px4_sitl_default/rootfs/etc/init.d-posix/px4-rc.simulator: gz_bridge: not found
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
desktop:~/PX4-Autopilot$ [Err] [SystemPaths.cc:534] File [/default.sdf] resolved to path [/default.sdf] but the path does not exist
[Err] [Server.cc:109] Failed to find world [/default.sdf]

monemati, 2024-03-03:
@Mojgan-arch Hello, I hope you are well. If you are using WSL you can check this out: https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html
You are getting the same error as #1 issue.

Mojgan-arch, 2024-03-03:
Thanks for prompt reply, my system is Ubuntu 22.4. 
since I use Ubuntu, in your opinion, what is the problem?

May I ask you to have your email, or connect to my email (mojgan.rezakhani8683@gmail.com)

monemati, 2024-03-03:
@Mojgan-arch Check your inbox. In the meantime check if your model path is set correctly and there is access to the drone model. Check "Additional Configs" in readme.

closed 2024-05-11 by monemati