monemati/PX4-ROS2-Gazebo-YOLOv8 #5: The problem of running simulation

GLtangzero, 2024-04-05:
Dear sir:
I have the problem of running the gazebo:
~/PX4-Autopilot$ PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 310 bytes, decoded 310 bytes (INT32:13, FLOAT:3)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/tang/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 268.08,-128.22,3.86,0.00,0,-0.7
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 268.08,-128.22,3.86,0.00,0,-0.7
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[Err] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
[Err] [SDFFeatures.cc:843] The geometry element of collision [collision] couldn't be created
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
tang@tang-GL502VSK:~/PX4-Autopilot$ [Err] [FuelClient.cc:691] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/sonoma raceway/2/sonoma raceway.zip
  REST response code: 504
[Err] [SystemPaths.cc:432] URI [https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sonoma raceway/2/files/meshes/Raceway.obj] resolved to path [/meshes/Raceway.obj] but the path does not exist
[Err] [SystemPaths.cc:525] Could not resolve file [https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sonoma raceway/2/files/meshes/Raceway.obj]
[Err] [MeshManager.cc:193] Unable to find file[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sonoma raceway/2/files/meshes/Raceway.obj]

And after that the GAZEBO could show the drone and car but can't show the world and it also can't connect to the MicroDDS and QgroundControl.

And also I can't running ros2 run ros_gz_image image_bridge /camera it show like:
$ ros2 run ros_gz_image image_bridge /camera
Package 'ros_gz_image' not found

Thanks for your model it looks very good, i really wish to running it.

monemati, 2024-04-05:
@GLtangzero Hello there. In your first problem it looks like the raceway model can not be downloaded from server, you can try to download the model from this [link](https://app.gazebosim.org/OpenRobotics/fuel/models/Sonoma%20Raceway) and put it in right directory in .gz model path.

For your second problem you can try the MicroDDS "v2.4.1" version (git checkout v2.4.1) and try 97cb933cff6d837e59dd55123804cd78abdca22a tag for PX4-Autopilot (git checkout 97cb933cff6d837e59dd55123804cd78abdca22a)

For your last problem you can try this command:
`sudo apt install ros-humble-ros-gzgarden` 
if it is not fixed reinstall ros 2 humble again.

GLtangzero, 2024-04-06:
Dear sir:
First of all, thanks for your reply, after I download the new model it could be show in the gazebo.
[image]

But the drone still can't take off, the QgroundControl show like this 
(1)Compass sensor 0 missing
(2)No valid local position estimate
(3)No manual control input
[image]

And if I  change the world to the origin PX4 default world the drone could be control, so I guess it's something problem happened in the world sdf document.

The camera could be used.

But control can't work:
(1)pygame 2.5.2 (SDL 2.28.2, Python 3.10.12)
Hello from the pygame community. https://www.pygame.org/contribute.html
Waiting for drone to connect...
-- Connected to drone!
Task exception was never retrieved
future: <Task finished name='Task-3' coro=<getKeyboardInput() done, defined at /home/tang/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py:12> exception=ActionError(<mavsdk.action.ActionResult object at 0x7e1ad339ad40>, 'arm()')>
Traceback (most recent call last):
  File "/home/tang/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py", line 36, in getKeyboardInput
    await my_drone.action.arm()
  File "/home/tang/px4-venv/lib/python3.10/site-packages/mavsdk/action.py", line 328, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()

(2)ros2 run px4_ros_com offboard_control
Starting offboard control node...
[INFO] [1712381251.842813578] [offboard_control]: Arm command send
[image]

monemati, 2024-04-06:
@GLtangzero Hello again. I did not try QGC back then with this repo. You said it can run with the default world so you can check the differences, for example in line 15 there is 
`<plugin name='gz::sim::systems::NavSat' filename='gz-sim-navsat-system'/>`
which I do not have in my sdf document.

And another thing is that my PX4-Autopilot version is a little bit old, maybe there is a miscommunication with the newer QGC.

GLtangzero, 2024-04-07:
@monemati 
Thank you for your help sir.
My px4-Autopilot is the newest type, I will try to check the problem.
If i get the resolve I will return again.
Thanks for your return again!

wyattPol, 2024-05-12:
> @monemati Thank you for your help sir. My px4-Autopilot is the newest type, I will try to check the problem. If i get the resolve I will return again. Thanks for your return again!

hey I have the same problem with you. Did you find any solution?

GLtangzero, 2024-05-13:
@wyattPol First please delete all model which you using in the gazebo, open the world/default.sdf and delete all     <include>
      <pose>295.96 -146.5 2.48 -0.019 -0.008 2.46</pose>
      <uri>model://pickup</uri>
      <name>pickup_2</name>
    </include> this type code and then run the progress again, if you can open it, then add the model one by one, I think is the problem of the model with gazebo and px4, please check it.

AlbertoRemus, 2024-06-25:
Hi! I think the problem is this missing part:

```
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <world_frame_orientation>ENU</world_frame_orientation>
      <latitude_deg>47.397971057728974</latitude_deg>
      <longitude_deg> 8.546163739800146</longitude_deg>
      <elevation>0</elevation>
    </spherical_coordinates>
```

GLtangzero, 2024-06-26:
@AlbertoRemus 
Thanks, I will try later.

evellynnicole, 2024-07-04:
> @AlbertoRemus Obrigado, vou tentar mais tarde.

I don't understand how you solved the control problem, I have the same problem:

Task exception was never retrieved
future: <Task finished name='Task-3' coro=<getKeyboardInput() done, defined at /home/evellyn/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py:12> exception=ActionError(< mavsdk.action.ActionResult object at 0x7b0de69f3730>, 'arm()')>
Traceback (most recent call last):
 File "/home/evellyn/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py", line 36, in getKeyboardInput
 await my_drone.action.arm()
 File "/home/evellyn/.local/lib/python3.10/site-packages/mavsdk/action.py", line 336, in arm
 raise ActionError(result, "arm()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params:()

Also, I can't view the drone's camera.

In terminal 2, I have this problem:
WARN [commander] Arming denied: Resolve system health failures first

GLtangzero, 2024-07-05:
@evellynnicole 
Hello, please try to use the below solution:
<1>Please check which gazebo you used in, if it's gazebo classic it will need QControground software to help arm, and if it's gazebo garden you need check the drone type. 
<2>Input `commander takeoff` in your Px4 terminal, than try to move the drone.
If anyother problem there please report detail and add some pictures.
Thanks!

evellynnicole, 2024-07-05:
I installed QGroundControl, now how do I solve the armament problems?

[image]

GLtangzero, 2024-07-06:
@evellynnicole 
Hello, sorry for repeat late.
Please go to the main page.
[image]
click the arm bottom and  drag the arm bar to arm the drone.

101ash, 2024-08-22:
@monemati Dear sirI hope it won't cause you any trouble.:
1.I have the problem of running in"Additional Configs" **in step of"Copy default.sdf from the worlds folder in the main repo to ~/PX4-Autopilot/Tools/simulation/gz/worlds/**",
my _default.sdf_ location in the PX4-Autopilot/Tools/simulation/gz/worlds/,I not sure that I need to copy again?
[image]
2.And I am not sure  my doing is correct or not,In step of "**Copy the content of models from main repo to ~/.gz/models**" I just copy the file of models from PX4-Autopilot/Tools/simulation/gz/models to.gz file.
Can you tell me how to do it correctly?
3.Then I use `nano ~/.bashrc and find `source /opt/ros/humble/setupbash`already on the bottom.so I need past it again?

monemati, 2024-08-22:
@101ash Hello.
1- Yes, copy `default.sdf` again, I did some modification and you need to replace `default.sdf` from my repo to `~/PX4-Autopilot/Tools/simulation/gz/worlds/` in your machine.
2- Navigate to `~/.gz/models` in your machine and just paste folders from my repo (PX4-ROS2-Gazebo-YOLOv8/models) into that folder, so you will have `~/.gz/models/casual_female` , `~/.gz/models/hatchback`, and so on, in your machine
3- If there is already a `source /opt/ros/humble/setup.bash` in your bashrc, you don't need to paste it again.

martinkeith, 2024-09-17:
> @monemati Thank you for your help sir. My px4-Autopilot is the newest type, I will try to check the problem. If i get the resolve I will return again. Thanks for your return again!

Hi sir，I  have the same problem,Do you resolve this problem in any way?

martinkeith, 2024-09-18:
> Dear sir: First of all, thanks for your reply, after I download the new model it could be show in the gazebo. ![Screenshot from 2024-04-06 14-13-14](https://private-user-images.githubusercontent.com/135005974/320184311-1867c52c-1b6c-4041-a56e-58d1404d375b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MzU1OTMsIm5iZiI6MTcyNjYzNTI5MywicGF0aCI6Ii8xMzUwMDU5NzQvMzIwMTg0MzExLTE4NjdjNTJjLTFiNmMtNDA0MS1hNTZlLTU4ZDE0MDRkMzc1Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOFQwNDU0NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYTZkZWEzNzdmYWEzOTBiZTFkN2FhZTkwMmJiNTc1Njk2NGYwZmQzMDYzODQ3YjUxYTVmN2U2NGQ1YzRlZTY2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.zdEsmXo2wGop-bzHIoVxU7pWpKVRJyGMEY_NYBQ9Gfg)
> 
> But the drone still can't take off, the QgroundControl show like this (1)Compass sensor 0 missing (2)No valid local position estimate (3)No manual control input ![Screenshot from 2024-04-06 14-24-47](https://private-user-images.githubusercontent.com/135005974/320184318-0f20bade-8ee7-4edb-a359-e01beb0cff15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MzU1OTMsIm5iZiI6MTcyNjYzNTI5MywicGF0aCI6Ii8xMzUwMDU5NzQvMzIwMTg0MzE4LTBmMjBiYWRlLThlZTctNGVkYi1hMzU5LWUwMWJlYjBjZmYxNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOFQwNDU0NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZGE1NWI5YzQ2ZjMyMjI1MDY2Y2JjZGQ1ZDJiZDg5ZjNiNmRmZWY1NmQ5MjBlOGQyMzIyYTBlNGJlOGEwMmRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.KaTO6maFsPCShkOMqz9gFUPAOJNfHxqrBvqPz8sS3i4)
> 
> And if I change the world to the origin PX4 default world the drone could be control, so I guess it's something problem happened in the world sdf document.
> 
> The camera could be used.
> 
> But control can't work: (1)pygame 2.5.2 (SDL 2.28.2, Python 3.10.12) Hello from the pygame community. https://www.pygame.org/contribute.html Waiting for drone to connect... -- Connected to drone! Task exception was never retrieved future: <Task finished name='Task-3' coro=<getKeyboardInput() done, defined at /home/tang/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py:12> exception=ActionError(<mavsdk.action.ActionResult object at 0x7e1ad339ad40>, 'arm()')> Traceback (most recent call last): File "/home/tang/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py", line 36, in getKeyboardInput await my_drone.action.arm() File "/home/tang/px4-venv/lib/python3.10/site-packages/mavsdk/action.py", line 328, in arm raise ActionError(result, "arm()") mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()
> 
> (2)ros2 run px4_ros_com offboard_control Starting offboard control node... [INFO] [1712381251.842813578] [offboard_control]: Arm command send ![Screenshot from 2024-04-06 14-28-12](https://private-user-images.githubusercontent.com/135005974/320184331-8946b091-584b-4f12-9425-4aa725fc7118.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MzU1OTMsIm5iZiI6MTcyNjYzNTI5MywicGF0aCI6Ii8xMzUwMDU5NzQvMzIwMTg0MzMxLTg5NDZiMDkxLTU4NGItNGYxMi05NDI1LTRhYTcyNWZjNzExOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOFQwNDU0NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDMxMTE0NWI1ZDIyOWRlYTg1NDQ4ZDc0ZTM0YjMyM2YzZGE4YmM2YmJmY2I4YWQ1YzQwN2Q0OTA4YzhmNDI5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2xjEDuycq-GLon8GEBBSg8nHA9Bo7R2W8_Rchy3oZt0)
Hi sir，I have the same problem,Do you resolve this problem in any way

martinkeith, 2024-09-18:
> @wyattPol First please delete all model which you using in the gazebo, open the world/default.sdf and delete all
> 
> ![Screenshot from 2024-05-13 13-51-53](https://private-user-images.githubusercontent.com/135005974/329926798-71642a6d-4db5-4a1d-ad37-fce20fefb399.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2MzU1OTMsIm5iZiI6MTcyNjYzNTI5MywicGF0aCI6Ii8xMzUwMDU5NzQvMzI5OTI2Nzk4LTcxNjQyYTZkLTRkYjUtNGExZC1hZDM3LWZjZTIwZmVmYjM5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxOFQwNDU0NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYTAxMTRjOTgxMGMzZWJmNzBlMzdkYWZhNmZjMWQ3ODhhZTg4OTZiODc4Zjk3YmE0MzE2ZGY3YmViMzdiNDliJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7hJXr5eggH-yPbsSS_8ojjtkhiYERvuxViAJ25lFILg)
> 
> this type code and then run the progress again, if you can open it, then add the model one by one, I think is the problem of the model with gazebo and px4, please check it.

I did what you said, first delete all files in .gz/models,then delete all<include> </include> as you mentioned. But failed,it show 'No valid local position estimate   No valide global position estimate' .   The GPS status info is GPS Count: 0 GPS Lock: None HDOP: --.-- VDOP:--.-- Course OverGround:--.--.   
So Did I do the right? Looking forward to your reply

martinkeith, 2024-09-18:
@monemati Hi bro. As mentioned before, I failed.  Then I used the original default.sdf instead of the default.sdf you provided. Run it, all fault is gone.  so if world/default.sdf you provided have something wrong?  Looking forward to your reply.

monemati, 2024-09-23:
@martinkeith Hello. I checked the latest [default.sdf](https://github.com/PX4/PX4-gazebo-models/blob/536305adee09b9ace391b16107e625cf7c6db7e7/worlds/default.sdf) from PX4 and apparently they changed some stuff (like plugins). I am not sure which differences are causing the problems, but when I got the time I will look into it.
(The PX4 version I used for this repo was `v1.14.0-beta2`)

closed 2024-12-22 by monemati