monemati/PX4-ROS2-Gazebo-YOLOv8 #6: Can't run camera topic

BJTUpupil, 2024-05-04:
when i use  ros2 run ros_gz_image image_bridge /camera to open the camear, but failed, the screen is empty, i inspect the topic list, it's does't have /camera topic. What 's the problem ?
```shell
ros2 topic list
/fmu/in/obstacle_distance
/fmu/in/offboard_control_mode
/fmu/in/onboard_computer_status
/fmu/in/sensor_optical_flow
/fmu/in/telemetry_status
/fmu/in/trajectory_setpoint
/fmu/in/vehicle_attitude_setpoint
/fmu/in/vehicle_command
/fmu/in/vehicle_mocap_odometry
/fmu/in/vehicle_rates_setpoint
/fmu/in/vehicle_trajectory_bezier
/fmu/in/vehicle_trajectory_waypoint
/fmu/in/vehicle_visual_odometry
/fmu/out/failsafe_flags
/fmu/out/position_setpoint_triplet
/fmu/out/sensor_combined
/fmu/out/timesync_status
/fmu/out/vehicle_attitude
/fmu/out/vehicle_control_mode
/fmu/out/vehicle_global_position
/fmu/out/vehicle_gps_position
/fmu/out/vehicle_local_position
/fmu/out/vehicle_odometry
/fmu/out/vehicle_status
/parameter_events
/rosout

```

monemati, 2024-05-06:
@BJTUpupil Hello, Double check if ros-humble-ros-gzgarden is installed by running sudo apt install ros-humble-ros-gzgarden, If it did not work do the "Install ROS 2" part in readme again.

BJTUpupil, 2024-05-07:
@monemati Hello, i run make px4_sitl gz_x500_depth have some problem, i got this:
```
make px4_sitl gz_x500_depth
[0/1] cd /home/kervin/project/drone-st...topilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 251 bytes, decoded 251 bytes (INT32:11, FLOAT:2)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/kervin/project/drone-stml/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 0,0,0,0,0,0
INFO  [gz_bridge] Model position z is less or equal 0.0, moving upwards
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-05-07/02_07_20.ulg	
INFO  [logger] Opened full log file: ./log/2024-05-07/02_07_20.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
``` 
Is it because of a problem with libEGL?
```
libEGL warning: egl: failed to create dri2 screen
```
Is it necessary to use an nvidia graphics card for rendering? The computer I use is a 4060+13700k notebook. I feel that it may be because the core graphics is always involved in the calculation. I have tried to fix this problem for several days, but to no avail. success

monemati, 2024-05-07:
@BJTUpupil Hi, the libEGL is only a warning. You are getting `INFO  [px4] Startup script returned successfully` and it means px4 is ok, what is your problem? You can't see the gazebo world or something else?

BJTUpupil, 2024-05-08:
@monemati Thank you very much for your answer. I installed it on a new system and found that some previous problems caused me to run python uav_camera_det.py unable get camera image, I will close this issue. 
Thanks again.

closed 2024-05-08 by BJTUpupil

lucas-muller-1998, 2025-07-22:
hello , i had the same problem 
can you tell me what the problem was ? everything is ok except when running uav_camera_det.py script no image is shown