monemati/PX4-ROS2-Gazebo-YOLOv8 #10: Gazebo sim window can't appear

wyattPol, 2024-05-10:
in the second terminal I run this command:~/PX4-Autopilot$ PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

the gazebo sim window appeared only for one time when I first time to run the command,but later I run this command, the window still didn't show up. I ve tried to use this command :pkill -f gazebo ,there is nothing changed. Here below is the terminal displays:
`(px4-venv) (mytello) fisher@flyingff2083:~/PX4-Autopilot$ PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 473 bytes, decoded 473 bytes (INT32:15, FLOAT:8)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] gazebo already running world: default
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 268.08,-128.22,3.86,0.00,0,-0.7
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 268.08,-128.22,3.86,0.00,0,-0.7
INFO  [lockstep_scheduler] setting initial absolute time to 545960000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-05-11/01_23_08.ulg  
INFO  [logger] Opened full log file: ./log/2024-05-11/01_23_08.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
INFO  [uxrce_dds_client] synchronized with time offset 1715390044610902us
INFO  [uxrce_dds_client] successfully created rt/fmu/out/battery_status data writer, topic id: 19
INFO  [uxrce_dds_client] successfully created rt/fmu/out/estimator_status_flags data writer, topic id: 87
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 93
INFO  [uxrce_dds_client] successfully created rt/fmu/out/manual_control_setpoint data writer, topic id: 141
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 180INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 203
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 226
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data

 writer, topic id: 242
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 249
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 256
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 261
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 266
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged

PX4 Exiting...
pxh> Exiting NOW.
`

monemati, 2024-05-11:
@wyattPol Hello. It says `INFO [init] gazebo already running world: default` just close everything and/or reboot to be sure.

wyattPol, 2024-05-11:
> @wyattPol Hello. It says `INFO [init] gazebo already running world: default` just close everything and/or reboot to be sure.

hi thanks for your help. It works now, but in the gazebo window, there is no drones, only the world. what could be the reason?here is my x500_depth model.sdf content:<?xml version="1.0" encoding="UTF-8"?>
<sdf version='1.9'>
  <model name='x500_depth'>
    <include merge='true'>
      <uri>x500</uri>
    </include>
    <include merge='true'>
      <uri>model://OakD-Lite</uri>
      <pose>.15 .029 .21 0 0.7854 0</pose>
    </include>
    <joint name="CameraJoint" type="fixed">
      <parent>base_link</parent>
      <child>OakD-Lite/base_link</child>
      <pose relative_to="base_link">.12 .03 .242 0 0 0</pose>
    </joint>
  </model>
</sdf>

monemati, 2024-05-13:
@wyattPol Hi, The offboard control is only an example, and if you look at the code from their [repo](https://github.com/PX4/px4_ros_com) you will see it is only takeoff and land. For keyboard control you can use `Fly using Keyboard` part of repo. If you want to use a ros node to run other commands than takeoff you have to modify px4_ros_com package (~/ws_offboard_control/src/px4_ros_com/src/examples).

wyattPol, 2024-05-17:
> @wyattPol Hi, The offboard control is only an example, and if you look at the code from their [repo](https://github.com/PX4/px4_ros_com) you will see it is only takeoff and land. For keyboard [...]

Thanks for your help!

closed 2024-05-17 by wyattPol