monemati/PX4-ROS2-Gazebo-YOLOv8 #9: error when launch PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

wyattPol, 2024-05-10:
i run this command in terminal 2 :Terminal #2:
cd ~/PX4-Autopilot
PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4 ,it give me error:
[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[0]/uri:/home/fisher/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf:L148]: Msg: Unable to find uri[model://sonoma_raceway]
[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[1]/uri:/home/fisher/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf:L152]: Msg: Unable to find uri[model://hatchback_blue]
[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[2]/uri:/home/fisher/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf:L157]: Msg: Unable to find uri[model://pickup]
[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[3]/uri:/home/fisher/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf:L162]: Msg: Unable to find uri[model://pickup]
[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[4]/uri:/home/fisher/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf:L167]: Msg: Unable to find uri[model://casual_female]
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256

i already followed the instruction to change the bashrc and copy models to .gz. please give some advice

monemati, 2024-05-10:
@wyattPol Hello. First:
Put below lines in your bashrc:
```
source /opt/ros/humble/setup.bash
export GZ_SIM_RESOURCE_PATH=~/.gz/models
```
And then run this command again to be sure:
`source ~/.bashrc`
Second: Check the structure of models' folder, for example it should be ~/.gz/models/sonoma_raceway and not ~/.gz/models/models/sonoma_raceway

wyattPol, 2024-05-10:
> @wyattPol Hello. First: Put below lines in your bashrc:
>
> ```
> source /opt/ros/humble/setup.bash
> export GZ_SIM_RESOURCE_PATH=~/.gz/models
> ```
>
> And then run this command again to be sure: `source [...]

hi! thank you for your help. i have  source /opt/ros/humble/setup.bash
export GZ_SIM_RESOURCE_PATH=~/.gz/models in my bashrc, also the model's folder structure is correct like you mentioned. but it still doesnt work. i dont know why....

wyattPol, 2024-05-10:
i should close all the old terminals and open new terminal,then it works,it can read models now.

closed 2024-05-10 by wyattPol