monemati/PX4-ROS2-Gazebo-YOLOv8 #15: Version confict during Install

ftondolo, 2024-12-19:
I am getting a version conflict during installation of ros-humble-ros-gzgarden. Specifically a version of the dependency ros-humble-ros-gz-sim is conflicting. Could I have some pointers as to what is going on?

monemati, 2024-12-22:
@ftondolo Hello. I think it is related to [EOL of gzgarden](https://community.gazebosim.org/t/gazebo-garden-officially-end-of-life/3278). You can try Harmonic version of Gazebo.

ftondolo, 2024-12-23:
You're right, I've resolved this by instead installing (what I believe is the Fortress version) the package ros-humble-ros-gz. However while I can now open the simulator and control the drone the uav_camera_det.py script gives no output other than the initial yolov8 weight download and the ros2 terminal outputs only "Unknown message type [9]."

closed 2024-12-23 by ftondolo