monemati/PX4-ROS2-Gazebo-YOLOv8 #16: rclpy.spin(image_subscriber) Hangs

ftondolo, 2024-12-23:
When running the simulator using (gazebo fortress instead of garden for EOL compatibility) I run into the issue that while I am able to fly the drone the line of code `rclpy.spin(image_subscriber)` hangs causing no frames to be run through the YoLov8 model. How could I resolve this?

ftondolo, 2024-12-23:
Moreover, the line `ros2 run ros_gz_image image_bridge /camera` outputs only Unknown message type [9].

monemati, 2024-12-23:
@ftondolo The `ros2 run ros_gz_image image_bridge /camera` bridges ros 1 drone's camera to ros 2, please check if the PX4 and MicroXRCE are running without issues. If there is no issues with drone's camera you can look in to [ros_gz](https://github.com/gazebosim/ros_gz) and build the compatible version based on your ROS 2 and Gazebo version from source.

closed 2024-12-23 by ftondolo