monemati/PX4-ROS2-Gazebo-YOLOv8 #18: how can impelement directly to my ros humble in pc without using the docer

ckjunaid, 2024-12-25:


monemati, 2024-12-25:
@ckjunaid Hi. You can follow `Installation` part from Readme. (It is right after docker)

ckjunaid, 2025-01-12:
Sir, I am actually going to fly a real drone of 10m and hover it on the same takeoff position without using gps and using ros humble ... Do u have any about the workflow of this

ckjunaid, 2025-01-12:
> @ckjunaid Hi. You can follow `Installation` part from Readme. (It is right after docker)

Hi

monemati, 2025-01-13:
@ckjunaid For real drone you need to take similar steps like the simulations, and the difference is configuring the real flight controller on your drone instead of using SITL. I'll show you some steps but you have to read tutorials in [PX4 docs](https://docs.px4.io/main/en/).
1. You should install PX4 firmware on the flight controller (like Pixhawk)
2. Install ROS2 on your computer
3. Install PX4-Autopilot on your computer
4. Install QGroundControl
5. Configure PX4 and drone (such as failsafes, flight modes, etc.)
6. Create ROS2 nodes (you can use examples [from docs](https://docs.px4.io/main/en/ros2/offboard_control.html))

Notes: 
- Look for offboard control in PX4 docs.
- In initial tests, do not use propellers on your rotors to be safe, use propellers after you are confident about the configurations.

monemati, 2025-01-16:
@ckjunaid That's because of failsafes, I know that flight controllers firmware is sensitive about having gps for a safe flight, but I'm not sure if you can fly the drone while gps is disconnected. You can look into failsafes and tune the gps_failsafe parameters to check if you can bypass it, but keep in mind that it's really dangerous to go without a connected gps in high altitude flights.