monemati/PX4-ROS2-Gazebo-YOLOv8 #19: Preflight check failed

TKPhuong, 2024-12-25:
[image]
I have followed each step of the installation (manual installation guide not the docker one) but still fail to fly the drone due to below preflight check fails. Everything else including gazebo simulation and the camera stream run without any problem.  
```
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station
```
	
which caused preflight check failed , therefore when send arm command it got denied as below
`WARN  [commander] Arming denied: Resolve system health failures first	`

I have tried docker installation and it worked, but because I want to change the world model to my custom world sdf and dont know how to do it in docker environment .
Does anyone encounter the same problem when install locally ?

monemati, 2024-12-25:
@TKPhuong Hi, I'm not sure why are getting these warnings, but you can try to use below params in PX4 SITL just to see if you can arm and fly the drone. (do not use these kind of commands in real life flight!)
```
param set COMPASS_ENABLE 0
param set ARMING_CHECK 0
```

And about the docker part: You can copy your default.sdf from your host machine into docker by using `docker cp` command in a new terminal: (change the address and container name based on your machine)
```
# Command template: docker cp HOST_ADDRESS CONTAINER_NAME:DOCKER_ADDRESS
docker cp ~/Desktop/default.sdf px4_ros2_gazebo_yolov8_container:/root/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
```

SunShiyue97, 2025-03-07:
Hi, i got the same problem and solved it with adding
 `<plugin name="gz::sim::systems::NavSat" filename="gz-sim-navsat-system"/>` 
to default.sdf

TKPhuong, 2025-03-19:
@monemati Thank you for replying. Settings parameters manually did not work for me.
@SunShiyue97 this indeed solved the problem. Thank you !