monemati/PX4-ROS2-Gazebo-YOLOv8 #20: COMMAND_DENIED when running keyboard-mavsdk-test.py

BINFLCY, 2024-12-31:
Hello, thank you for providing this great project! I encountered an issue when trying to run the keyboard-mavsdk-test.py script. When I press the r key to arm the drone, I receive a COMMAND_DENIED error. The full error log is as follows:
`zj@zj-Z390-UD:~/ws_offboard_control$ source ~/px4-venv/bin/activate
cd ~/PX4-ROS2-Gazebo-YOLOv8
python keyboard-mavsdk-test.py
pygame 2.6.1 (SDL 2.28.4, Python 3.10.12)
Hello from the pygame community. https://www.pygame.org/contribute.html
Waiting for drone to connect...
-- Connected to drone!
Task exception was never retrieved
future: <Task finished name='Task-3' coro=<getKeyboardInput() done, defined at /home/zj/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py:12> exception=ActionError(<mavsdk.action.ActionResult object at 0x74557f9c55a0>, 'arm()')>
Traceback (most recent call last):`
[image]

BINFLCY, 2024-12-31:
[image]

monemati, 2025-01-02:
@BINFLCY Hi, I am not sure if the problem is from MAVSDK part or the integration with QGC is causing the problem. Please try to do the process without QGC to check if script is working or not.

BINFLCY, 2025-01-03:
Thank you very much for your response and guidance. I truly appreciate your help—it means a lot to a beginner like me.

Following your suggestion, I tried controlling the system with the keyboard without using QGC. I found that the issue still occurs when pressing the "R" key, as it raises the error: raise ActionError(result, "arm()"). However, the "L" key works as expected, displaying "Flight Mode: LAND."

When using QGC, the "L" key also works properly, with the log showing: "landing at current position." This leads me to believe that the communication is functioning normally.

I also attempted the solution mentioned in [this issue](https://git.jl-k.com/monemati/PX4-ROS2-Gazebo-YOLOv8/issues/5), which suggests manually arming the vehicle via QGC. However, the arm command doesn’t seem to have any effect in my case.
[image]
[image]
[image]

monemati, 2025-01-03:
@BINFLCY If you already tried without QGC and the arming issue is still there, my next guess is you are using the new version of PX4 firmware which is v1.16 (I have v1.15 on my machine). You can try to use the v1.15 firmware version and see if everything is okay or not. You can also check [this issue](https://github.com/PX4/PX4-Autopilot/issues/24033) in PX4 repo.

BINFLCY, 2025-01-06:
Thank you so much for your patient guidance! I’m thrilled to share that I’ve successfully achieved flight—what an incredible moment! This is truly an amazing project, and I’m deeply impressed by the thought and effort you’ve put into it.

As part of the process, I made some modifications, such as downloading PX4 version 1.15.2 and editing the default.sdf file. Moving forward, I plan to incorporate a large model into the project to enable it to describe the scenes captured by the drone. I’m really excited about the possibilities ahead.

Once again, thank you for your remarkable contributions and dedication to open source. It’s truly inspiring!
`<?xml version="1.0" encoding="UTF-8"?>
<sdf version="1.9">
  <world name="default">
    <physics type="ode">
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1.0</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
    </physics>
    <plugin name="gz::sim::systems::Physics" filename="gz-sim-physics-system"/>
    <plugin name="gz::sim::systems::UserCommands" filename="gz-sim-user-commands-system"/>
    <plugin name="gz::sim::systems::SceneBroadcaster" filename="gz-sim-scene-broadcaster-system"/>
    <plugin name="gz::sim::systems::Contact" filename="gz-sim-contact-system"/>
    <plugin name="gz::sim::systems::Imu" filename="gz-sim-imu-system"/>
    <plugin name="gz::sim::systems::AirPressure" filename="gz-sim-air-pressure-system"/>
    <plugin name="gz::sim::systems::ApplyLinkWrench" filename="gz-sim-apply-link-wrench-system"/>
    <plugin name="gz::sim::systems::NavSat" filename="gz-sim-navsat-system"/>
    <plugin name="gz::sim::systems::Sensors" filename="gz-sim-sensors-system">
      <render_engine>ogre2</render_engine>
    </plugin>
    <gui fullscreen="false">
      
      <plugin filename="MinimalScene" name="3D View">
        <gz-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </gz-gui>
        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
        <camera_clip>
          <near>0.25</near>
          <far>25000</far>
        </camera_clip>
      </plugin>
      
      <plugin filename="EntityContextMenuPlugin" name="Entity context menu">
        <gz-gui>
          <property key="state" type="string">floating</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="GzSceneManager" name="Scene Manager">
        <gz-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="InteractiveViewControl" name="Interactive view control">
        <gz-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="CameraTracking" name="Camera Tracking">
[... 175 lines cut ...]
    </include>
    <include>
      <pose>273.43 -137.65 2.86 0.02 0 -0.65</pose>
      <uri>model://pickup</uri>
      <name>pickup_1</name>
    </include>
    <include>
      <pose>295.96 -146.5 2.48 -0.019 -0.008 2.46</pose>
      <uri>model://pickup</uri>
      <name>pickup_2</name>
    </include>
    <include>
      <pose>280.64 -137.32 2.77 -0.019 -0.008 2.46</pose>
      <uri>model://casual_female</uri>
    </include>    
    
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <world_frame_orientation>ENU</world_frame_orientation>
      <latitude_deg>47.397971057728974</latitude_deg>
      <longitude_deg> 8.546163739800146</longitude_deg>
      <elevation>0</elevation>
    </spherical_coordinates>
  </world>
</sdf>`

BINFLCY, 2025-01-06:
Thank you so much for your patient guidance! I’m thrilled to share that I’ve successfully achieved flight—what an incredible moment! This is truly an amazing project, and I’m deeply impressed by the thought and effort you’ve put into it.

As part of the process, I made some modifications, such as downloading PX4 version 1.15.2 and editing the default.sdf file. Moving forward, I plan to incorporate a large model into the project to enable it to describe the scenes captured by the drone. I’m really excited about the possibilities ahead.

Once again, thank you for your remarkable contributions and dedication to open source. It’s truly inspiring!

BINFLCY, 2025-01-06:
<?xml version="1.0" encoding="UTF-8"?>
<sdf version="1.9">
  <world name="default">
    <physics type="ode">
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1.0</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
    </physics>
    <plugin name="gz::sim::systems::Physics" filename="gz-sim-physics-system"/>
    <plugin name="gz::sim::systems::UserCommands" filename="gz-sim-user-commands-system"/>
    <plugin name="gz::sim::systems::SceneBroadcaster" filename="gz-sim-scene-broadcaster-system"/>
    <plugin name="gz::sim::systems::Contact" filename="gz-sim-contact-system"/>
    <plugin name="gz::sim::systems::Imu" filename="gz-sim-imu-system"/>
    <plugin name="gz::sim::systems::AirPressure" filename="gz-sim-air-pressure-system"/>
    <plugin name="gz::sim::systems::ApplyLinkWrench" filename="gz-sim-apply-link-wrench-system"/>
    <plugin name="gz::sim::systems::NavSat" filename="gz-sim-navsat-system"/>
    <plugin name="gz::sim::systems::Sensors" filename="gz-sim-sensors-system">
      <render_engine>ogre2</render_engine>
    </plugin>
    <gui fullscreen="false">
      
      <plugin filename="MinimalScene" name="3D View">
        <gz-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </gz-gui>
        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
        <camera_clip>
          <near>0.25</near>
          <far>25000</far>
        </camera_clip>
      </plugin>
      
      <plugin filename="EntityContextMenuPlugin" name="Entity context menu">
        <gz-gui>
          <property key="state" type="string">floating</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="GzSceneManager" name="Scene Manager">
        <gz-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="InteractiveViewControl" name="Interactive view control">
        <gz-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
          <property key="showTitleBar" type="bool">false</property>
        </gz-gui>
      </plugin>
      <plugin filename="CameraTracking" name="Camera Tracking">
        <gz-gui>
          <property key="resizable" type="bool">false</property>
          <property key="width" type="double">5</property>
          <property key="height" type="double">5</property>
          <property key="state" type="string">floating</property>
[... 170 lines cut ...]
    </include>
    <include>
      <pose>273.43 -137.65 2.86 0.02 0 -0.65</pose>
      <uri>model://pickup</uri>
      <name>pickup_1</name>
    </include>
    <include>
      <pose>295.96 -146.5 2.48 -0.019 -0.008 2.46</pose>
      <uri>model://pickup</uri>
      <name>pickup_2</name>
    </include>
    <include>
      <pose>280.64 -137.32 2.77 -0.019 -0.008 2.46</pose>
      <uri>model://casual_female</uri>
    </include>    
    
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <world_frame_orientation>ENU</world_frame_orientation>
      <latitude_deg>47.397971057728974</latitude_deg>
      <longitude_deg> 8.546163739800146</longitude_deg>
      <elevation>0</elevation>
    </spherical_coordinates>
  </world>
</sdf>

BINFLCY, 2025-01-06:
[Uploading default.txt…]()

3odat, 2025-04-03:
I have the same problem. is someone can help me in my project that would be appreciated. I just need it on my local Server without using a docker. can someone help me with that if that is possible  @monemati @BINFLCY

eOvic, 2025-07-31:
@monemati @hong3nguyen @3odat i fixed the Issue you just have to add this line to default.sdf : 

<plugin name="gz::sim::systems::NavSat" filename="gz-sim-navsat-system"/>

Hoolo, 2025-09-30:
I had to add the following to the world file as well, as for some reason even with the Navsat plugin my magnetometer data was not being published
`<plugin name="gz::sim::systems::Magnetometer" filename="gz-sim-magnetometer-system"/>`