monemati/PX4-ROS2-Gazebo-YOLOv8 #21: PX4 Gazebo startup error on first execution due to gz_bridge timeout

zahrakhan010, 2025-01-07:
When executing the following command in the terminal:

PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4

The system throws an error with the following output:

INFO  [init] starting gazebo with world: /home/coe/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256

The world file is located in ~/PX4-Autopilot/Tools/simulation/gz/worlds, and I have set the environment variable GZ_SIM_RESOURCE_PATH in my .bashrc:

export GZ_SIM_RESOURCE_PATH=~/.gz/models

Gazebo opens with the warehouse world, but the drone does not spawn until I re-run the command. The error seems to occur only during the first execution. Any suggestions or ideas on what might be causing this issue?

zahrakhan010, 2025-01-14:
thank you for giving me suggestions, resolve the above bug, but now again im facing few issues, in terminal 2 run this command PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="283.08,-136.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4
 and gave errors : " _____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 442 bytes, decoded 442 bytes (INT32:16, FLOAT:6)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/coe/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 0,0,1,0,0,0
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 0,0,1,0,0,0
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
WARN  [gz_bridge] Service call timed out as Gazebo has not been detected.
INFO  [lockstep_scheduler] setting initial absolute time to 104000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-01-14/06_16_37.ulg	
INFO  [logger] Opened full log file: ./log/2025-01-14/06_16_37.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [tone_alarm] notify positive
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [uxrce_dds_client] synchronized with time offset 1736835416087409us
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station	
INFO  [uxrce_dds_client] successfully created rt/fmu/out/battery_status data writer, topic id: 19
INFO  [uxrce_dds_client] successfully created rt/fmu/out/estimator_status_flags data writer, topic id: 82
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 88
INFO  [uxrce_dds_client] successfully created rt/fmu/out/manual_control_setpoint data writer, topic id: 137
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 181INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 213
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 236
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_land_detected data writer, topic id: 265
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 252
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 259
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 260
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_gps_position data writer, topic id: 262
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 266
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 271
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 276
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged

PX4 Exiting...
pxh> Exiting NOW.
"  and in Terminal 5, I ran the command:

ros2 run px4_ros_com offboard_control 

The output from Terminal 5 was:

coe@coe-System-Product-Name:~/ws_offboard_control$ ros2 run px4_ros_com offboard_control
Starting offboard control node...
[INFO] [1736834613.853484796] [offboard_control]: Arm command send

Although the arm command was successfully sent, the drone does not fly. Please help me resolve these issues and debug the problem.

monemati, 2025-01-16:
@zahrakhan010 At first try PX4_GZ_STANDALONE=0 parameter like below:
`PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="283.08,-136.22,3.86,0.00,0,-0.7" PX4_GZ_STANDALONE=0 PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4`
If it did not fixed the issue then there is possible problem about time sync of xrce which you can follow [this link](https://docs.px4.io/main/en/ros/ros2_comm.html#ros-gazebo-and-px4-time-synchronization). It's basically saying that you need to turn off `UXRCE_DDS_SYNCT`.

zahrakhan010, 2025-01-20:
Since i'm new to ROS2, and i'm having difficulty figuring out how to bridge the LiDAR data to ROS2.

monemati, 2025-01-20:
@zahrakhan010 I have not done that myself, but I think for lidar you should use a [version of x500 drone](https://docs.px4.io/main/en/sim_gazebo_gz/#running-the-simulation) which has lidar on it, and if there is a need for bridging, you can use [ros_gz_bridge](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_bridge), something like below command (change lidar topic name, and type of message to pointcloud2 if needed):
`ros2 run ros_gz_bridge parameter_bridge /lidar_scan@sensor_msgs/msg/LaserScan[sensor_msgs/LaserScan`

zahrakhan010, 2025-01-28:
Thank you so much for your help in resolving the previous issues I was facing. I really appreciate it!

Now, I have two concerns:

1.     QGroundControl is unable to read distance sensor data. I have set up the LiDAR sensor in Gazebo simulation, but the distance data is not being displayed in QGroundControl, however /lidar_scan topic successfully displayed sensor data. How can I resolve this issue?

2.     Issue with ros2 topic echo. When I run ros2 topic list, it shows many topics. However, when I try to echo a topic using:

ros2 topic echo /fmu/in/vehicle_visual_odometry

I get the following error:

    The message type 'px4_msgs/msg/VehicleOdometry' is invalid.

    How can I fix this issue?

Could you please help me resolve these problems?