monemati/PX4-ROS2-Gazebo-YOLOv8 #25: Cannot ARM

3odat, 2025-04-03:
Thank you for this really great Job!

If you would not mind to help me to setup the environment on Linux 22.04 because i followed all of your tutorial just the last step is cannot arm and control the drone motion?

Im working on research and i need an environment ASAP!

monemati, 2025-04-05:
@3odat Hi! Can you share the errors you are facing?

Side note: I suggest you to try the same version of PX4 (v1.15.0) which I used, to check if everything is fine, then you can try the newer versions of PX4.

ShubhangiSRG24, 2025-04-23:
Thank you for this page and detailed documentation!

I am facing the similar issue @3odat is facing to arm and control the drone motion. I am unable to get the drone to move only at the last step. I am working on a research, it would be great if we could get some help with this error. Thank you!

The error from the last terminal to arm and control drone is as follows:'~/PX4-ROS2-Gazebo-YOLOv8$ python keyboard-mavsdk-test.py
pygame 2.6.1 (SDL 2.28.4, Python 3.10.12)
Hello from the pygame community. https://www.pygame.org/contribute.html
Waiting for drone to connect...
-- Connected to drone!
Task exception was never retrieved
future: <Task finished name='Task-3' coro=<getKeyboardInput() done, defined at /PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py:12> exception=ActionError(<mavsdk.action.ActionResult object at 0x7f91621966e0>, 'arm()')>
Traceback (most recent call last):
  File "/PX4-ROS2-Gazebo-YOLOv8/keyboard-mavsdk-test.py", line 36, in getKeyboardInput
    await my_drone.action.arm()
  File "/px4-venv/lib/python3.10/site-packages/mavsdk/action.py", line 336, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()
'
And this is the PX4 terminal logs:'INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 393 bytes, decoded 393 bytes (INT32:16, FLOAT:4)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/young/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
INFO  [init] PX4_GZ_MODEL_POSE set, spawning at: 268.08,-128.22,3.86,0.00,0,-0.7
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 268.08,-128.22,3.86,0.00,0,-0.7
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
WARN  [vehicle_angular_velocity] no gyro selected, using sensor_gyro:0 1310988
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2025-04-23/09_52_54.ulg	
INFO  [logger] Opened full log file: ./log/2025-04-23/09_52_54.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [uxrce_dds_client] synchronized with time offset 1745401974245663us
INFO  [uxrce_dds_client] successfully created rt/fmu/out/battery_status data writer, topic id: 19
INFO  [uxrce_dds_client] successfully created rt/fmu/out/estimator_status_flags data writer, topic id: 87
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 93
INFO  [uxrce_dds_client] successfully created rt/fmu/out/manual_control_setpoint data writer, topic id: 141
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 180INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 203
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 226
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 242
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 249
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 256
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 261
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 266
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_command_ack data writer, topic id: 246
WARN  [commander] Arming denied: Resolve system health failures first	
INFO  [tone_alarm] notify negative
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged
'