monemati/PX4-ROS2-Gazebo-YOLOv8 #34: Camera and ROS2 Jazzy

eOvic, 2025-07-26:
Hello I'm using ubuntu 24.04 
ROS2 Jazzy, PX4 latest version and Gazebo harmonic 
When I try to run the script to use uav_camera.py it doesn't work and i keep waiting and nothing happens. 
PS : ros2 topic list doesn't provide a topic with /camera

Thanks for the help.

eOvic, 2025-07-26:
follow up, to see the camera topics you need to change the sdf file and run ros2 run ros_gz_bridge parameter_bridge \
/depth_camera@sensor_msgs/msg/Image@gz.msgs.Image \
/depth_camera/points@sensor_msgs/msg/PointCloud2@gz.msgs.PointCloudPacked
in a terminal everytime so Gazebo topics connect to ROS topics 

is this correct ?

monemati, 2025-07-26:
@eOvic Hi, the camera topic should be bridged without making changes in the sdf file, The reason you can't access to /camera topic might be related to the latest changes in PX4. The latest working combination I have in my machine is PX4 v1.15 + Gazebo Harmonic. You can also try to use docker and check if everything is fine or not.